jpraus/arduino-opentherm #2: Hardware fallback in case MUC is non-operational

renne, 2018-04-13:
Section 3.5 of the [OpenTherm Protocol Specification v2.2](http://www.domoticaforum.eu/uploaded/Ard%20M/Opentherm%20Protocol%20v2-2.pdf) defines a short-circuit of the OpenTherm interface as boiler enable. I suggest to short-circuit the OT-interface in case the microcontroller is non-operational to ensure availability of heating.

jpraus, 2018-04-16:
Thank you for your feedback.

Specification defines that master should short circuit wires in case it wont receive valid response within 5s. This is for case when boiler does not support OT+/OT- protocol at all. Since this library is desired to be used for custom solutions boiler should always support OT otherwise there is no need for use of library.

Have I understand your request correctly? Let me know.

renne, 2018-04-17:
I'm referring to the situation the microprocessor or power-supply of the microprocessor of the master circuit fails during heating period. In that case the master circuit should short-circuit the OpenTherm data connections to ensure basic function of the heating system to avoid frost damage.

jpraus, 2018-04-18:
In case master is not engaging communication with slave (thermostat or regulator is stuck or poweroff), slave should act as if no master is connected. For example if I disconnect my thermostat from boiler, boiler will heat on his own according to internal settings, same will work if mictrocontroller stop working. 

Even so, I dont see any way how to detect such malfunction of mictrocontroller and short-circuit Opentherm interface. You are welcome to suggest any working solution.

renne, 2018-04-18:
How about a GPIO of the the micro-controller as watchdog-logic and an opto-coupler as NOT-gate?

If the micro-controller is working correctly the input of the opto-coupler is active and the output is high-resistance. In case the micro-controller is not operational the input of the opto-coupler is inactive and the output low-resistance.

jpraus, 2018-05-03:
Your solution will work unless micro-controller is stuck. But again this situation should be handled by slave, if it does not receive a valid OT communication from master, it should behave as if master is not installed. In situation where slave does not support OT it makes no sense using this library.

Can you give me example where this would be useful?