jpraus/arduino-opentherm #4: Bug with f88 converter

senhtry, 2019-10-22:
Since OpenTherm protocol use the Two's complement (https://en.wikipedia.org/wiki/Two%27s_complement), it's necessary to minus 1 for negative number.

```
void OpenthermData::f88(float value) {
  if (value >= 0) {
    valueHB = (byte) value;
    float fraction = (value - valueHB);
    valueLB = fraction * 256.0;
  }
  else {
    valueHB = (byte)(value - 1);
    float fraction = (value - valueHB - 1);
    valueLB = fraction * 256.0;
  }
}
```

Example :  
A temperature of -5.25°C in f8.8 format is represented by the 2-byte value FAC0 hex 
(FAC0hex = - (10000hex-FAC0hex) = - 0540hex = - 1344dec, dividing by 256 gives -5.25) 

With your old code, you will get the wrong FBC0 hex instead of correct FAC0 hex.

jpraus, 2019-10-22:
Hi, thank you for finding this bug. Yeah, I've never really worked with negative temperatures. Should I incorporate your code or do you wish to create a pull request?

senhtry, 2019-10-22:
> Hi, thank you for finding this bug. Yeah, I've never really worked with negative temperatures. Should I incorporate your code or do you wish to create a pull request?

You can simply use my code :)

closed 2019-11-04 by jpraus