jpraus/arduino-opentherm #5: Possibly `Testing Hardware` section has some issues

yury-sannikov, 2019-12-26:
Hi Jiří, thank you for the great project!
A while ago I bought an OT GW shield and just recently tried to assemble it. What I found:
1. R5, R11 was 1k instead of 10k, no problem at all, just general feedback.
2. I found the `Thermostat interface` section is a bit misleading. 
```
If you re-connect MASTER-OUT pin to the ground again (same as in step 3 above),
blue RxB LED should go off. Inbound boiler communication is working.
```

If I connect MASTER-OUT pin to the ground, blue led stays on, but the green led is turning on. Which actually makes sense. When MASTER-OUT is grounded, SLAVE-IN should go up (therm and boiler terminals are interconnected).

The same for `If you connect SLAVE-OUT pin (digital pin 5 / sixth pin from the top on the left side) to +5V pin, the green RxT LED should light up`. Instead of `green RxT LED should light up`, I think docs should say, `blue RxB LED should go off`.

So, led behavior description should be swapped between item 5 and item 6.

Other parameters such as voltage and current match the description.

Let me know if I screwed something up and the docs are actually correct.

jpraus, 2020-01-01:
Hi, thank you for your feedback. You are totally right with the testing out hardware section. I've mixed up the green and blue values. It is fixed now. Thank you very much for your help and sorry for the inconvenience!

Regarding R5, R11 - you mean in the kit was wrong values ???

jpraus, 2020-01-01:
Commit: https://github.com/jpraus/arduino-opentherm/commit/9b69f47b1eec462fd95a076721980454f87e0c7c

yury-sannikov, 2020-01-02:
Cool, thank you very much!
As to the kit, R5 and R11 were 1k instead of 10k. Again, not a big deal, just FYI.

What I also noticed is that the voltage drop on D2 is around 4.2V instead of 4.7V. I tweaked R2 a bit to make U3 voltage closer to the datasheet values.

jpraus, 2020-01-03:
Cool! What value have you used for R2?

yury-sannikov, 2020-01-03:
lol, I was lazy so I added 4.7k in parallel to R2. So total resistance became 210ohm. 

I don't think that this is a schematic issue though. I suspect the D2 backward voltage nominal was different.
Frankly, I swapped all 3 diodes while unpacking so I made a quick test scheme as below. I tested D2 D3 D4 voltage drops. D3 was 13V as shown on the scheme (well, actually I read 12v because of the voltage source)
D2 and D4 shown different (lower) voltages. I don't remember exact readings though. I put the lowest one as D4, the next one as D3.

```
        10k
+12 -----===----------
             _|_     |
             / \     |
             ---    (V)
              |      |
gnd ------------------
```

I'm not good at electronics, also the scheme worked as-is so I thought those voltages are not required to be 100% eq to the schematic values.
I just corrected R2 to make U3 voltage drop closer to the datasheet.

closed 2020-01-03 by yury-sannikov