jpraus/arduino-opentherm #6: Testing with the 3.3 V version

yvdh, 2020-01-03:
Hello,

There is some information missing in the testing instructions:  basically they can only work in a standalone fashion in the 5 Volt configuration! For them to work in the 3.3 Volt configuration a 3.3 Volt power source is needed because this is not provided internally as is the case with the 5 Volt config. Basically the Q4 transistor base is not powered so MASTER OUT cannot work, and the same goes for the optocoupler for SLAVE IN.

So for testing **without a connected MCU** (= standalone), pins 3.3 V and 5 V on the right should be connected to each other (Obviously this will fry a  3.3 V MCU if it's connected, so be careful).

Mine now passes all the tests with an ESP32 board.

jpraus, 2020-01-07:
Yes! You are right and have a correct solution. Thank you!

closed 2020-02-18 by jpraus

sodd, 2023-02-06:
thx for the summarisation my problem how to test 3v3 logic. I was nervous about to power up esp32 from 5v and then connect 3v3 back to shield 3v3 pin.

yvdh, 2023-11-11:
Hello again.

Back in 2020 I never got my boiler to work with the OpenTherm shield (Remeha Quinta Solo): as soon as I connected it the boiler would start working full tilt.
I though this was a boiler issue, and dropped the project.
Recently I picked it up again and upon measuring the shield I find that the output voltage for your test #4 only reads 14.5 V instead of 15 - 18 V. According to OPenTherm specs this is too low to represent the 'high' state. Any suggestions? 
Remember I am testing with the 5V connected to the 3.3 V on the shield as I have modified it to work with a 3.3 V MCU ...
Cheers,
Yves