jpraus/arduino-opentherm #7: Wemos D1 R2

yury-sannikov, 2020-04-07:
Hi @jpraus 
Thanks again for the great stuff. I'm super slowly adopting it, lol.

I'm using WeMos D1 R2. I found that the pinout is a bit different. Basically D5 in R1 became D3 in R2.

That way Wemos D1 R2 Setup could be:
```
// Wemos D1 R2 Setup
//#define THERMOSTAT_IN 16
//#define THERMOSTAT_OUT 4
//#define BOILER_IN 5
//#define BOILER_OUT 0
```

jpraus, 2020-04-30:
Excellent, thank you! Will incorporate this into the code!

closed 2020-04-30 by jpraus

NijzinkM, 2020-04-30:
Pin 16 on ESP8266 is the only pin that does not support interrupts. Will `OPENTHERM::listen(THERMOSTAT_IN)` cause problems?

jpraus, 2020-05-01:
Interrupts are not needed, I rewrote the code some time ago to not use interrupts, it was causing lot of incompatibilities.

NijzinkM, 2020-05-01:
Thanks! Sorry for bothering.

jpraus, 2020-05-04:
Not at all!