jpraus/arduino-opentherm #9: Hardware Operation at 3.3V

DaniilMir, 2020-04-13:
Hello All,
My adapter board failed a self test with 3.3V mod. 
I started checking schematic and found the following : 
[image]
As per specification, input voltage on the slave Interface should be about 6V for the Low logic level and about 15V for the High logic level.
Resistors R8 and R11 Make a voltage divider. It divides the input voltage by 8 for 3.3V version.
Calculated output values are  1.87 for a High level and 0.74 for a low level.
I found a post with explanation regarding the ESP8266 logic levels : 
[http://henrysbench.capnfatz.com/henrys-bench/arduino-projects-tips-and-more](http://henrysbench.capnfatz.com/henrys-bench/arduino-projects-tips-and-more/esp8266ex-gpio-high-and-low-input-thresholds/)
On the right side of my picture you can see calculated logic levels for the ESP8266.
Low level value (0.74) fits there because it is below 0.825.
But a high level value (1.87) is below 2.47 threshold.
It looks like levels on this board are out of specs and are incompatible with ESP8266.
I think that it makes sense to make High level closer to 18V and low level to be closer to 5V by replacing zener diodes. 
And also use 6.2K instead of 4.7K for R11.
This way we will divider by ~6.3 
High level for the MCU will become 2.8 and low level will be 0.79 which fits into the specs.

NijzinkM, 2020-04-26:
I'm also using an EPS8266 board. Because of your issue, I decided to reproduce the problem. This is the Arduino code used for testing:
```
void loop() {
  if (digitalRead(2) == HIGH) {
    digitalWrite(LED_BUILTIN, HIGH);
  } else {
    digitalWrite(LED_BUILTIN, LOW);
  }
}
```
When I connected GPIO2 to 3.3V, the built-in LED turns off (HIGH turns the led off). When connected to GND, the LED turns on, as you would expect.

Then I created a 1.9V line by dividing the 3.3V with resistors. I connected GPIO2 to this 1.9V and the LED turned off. Connected to GND, LED turned on. Connected to 1.9V again, LED turned off. No floating pin state.

Although the HIGH threshold seems to be 2.47V according to the article you mentioned, the test results show that 1.9V also triggers high, so the issue might not be a problem in practice.

DaniilMir, 2020-04-26:
@NijzinkM Thank you for testing it on another board.

It may work, but let's think about repeatability of schematic and about parts tolerance. 
In university we did calculations in Matlab to decide which tolerance of parts may be used in which places of schematic to match specs. 
I ended with the following changes: 

1. D3 15V instead of 13V
2. D4 3.3V instead of 4.3V
3. R8 56K instead of 33K
4. R11 10K instead of 4.7K
5. R5 3.9K instead of 10K (schematic says that there should be 10K but board photos show 3.9K value)

This mod lowers the low value and increases the high logic value due to fit in the controller specs.
But still keeping voltage levels inside the OpenTherm voltage range.

Original schematic was designed to use the PIC controllers: 
[http://otgw.tclcode.com/features.html](http://otgw.tclcode.com/features.html)
Quote 1: 

> Another fortunate coincidence is that the output of the voltage reference module of the PIC uses the same pin that the comparators use as their common reference input. This means that the voltage reference module can easily be used to provide the threshold voltage for the comparators. This way the threshold voltage is not fixed in hardware, but can be made configurable through software.
The selected resistor values put the low level on the input of the first comparator below 0.87V and the high level between 1.87V and 2.24V. Using the low range of the voltage reference module the voltage can be specified in steps of around 0.2V. A value of 6 results in 1.25V, which falls nicely within the threshold values suggested in the OpenTherm specification. The threshold value for the second comparator is not very critical. A 1.25V threshold will do just nicely there too.

Quote 2:

> If you see nothing, or things like "Error 01" or "Error 03", the reference voltage may be wrong. You can modify it via Options → Settings (F5) or with the VR=X command. where X is a number between 0 and 9. The normal value is 1.250V (X=3), but due to variations in the used components you may have to change it. If you find you need a value less than 1.042V (X<2) or more than 1.667V (X>5), check that R5 and R6 have the right values. Don't forget to click the Set button after every adjustment to actually make the change in the gateway.

So by design they don't care about the threshold values, because it may be set in the firmware.
In our case we are not using analog input, so it would be nice follow the specs just to avoid possible strange hardware bugs.

NijzinkM, 2020-04-26:
That's very useful information. I agree that the current 3.3V modification in the readme is not reliable, although it might function properly. Thank you for supplying your resistor and diode values.

DaniilMir, 2020-04-26:
BTW
[http://ihormelnyk.com/opentherm_adapter](http://ihormelnyk.com/opentherm_adapter)
this project is using 15V zener as well.

jpraus, 2020-04-30:
Yes, the current 3.3V solution is a quick hack to reuse current hardware in order to allow 3.3V logic. However the new 3.3V board is coming soon, here is the updated schematics:
https://github.com/jpraus/arduino-opentherm/blob/master/doc/shield-schematic_rev4.png

jpraus, 2020-04-30:
> @NijzinkM Thank you for testing it on another board.
>
> It may work, but let's think about repeatability of schematic and about parts tolerance.
> In university we did calculations in Matlab to decide [...]

This is some really great stuff! Let me test you're setup out!

DaniilMir, 2020-05-01:
> Yes, the current 3.3V solution is a quick hack to reuse current hardware in order to allow 3.3V logic. However the new 3.3V board is coming soon, here is the updated schematics:
> https://github.com/jpraus/arduino-opentherm/blob/master/doc/shield-schematic_rev4.png

Great option. 
Q6 will definitely do the trick. Maybe R15 can have a bit higher value say 3.3V or 4.7K to reduce current consumption. But not higher than 12K

One more thing to add. As I can understand - Master Out is inverted.
If we apply high logic level on Master out OT has ~6mA current 
With low level on Master out  OT has ~20mA current 

So it is impossible to put jumpers between _master out_ -> _slave in_  and _slave out_ -> _master in_
just to run this board without MCU attached. (in case of MCU failure or for test purposes)

Q1, Q2, R6 and R7 is a constant current generator. Like [https://www.electroschematics.com/led-constant-current/](https://www.electroschematics.com/led-constant-current/)
 Q3 in fact acts as a switch to add R9 in parallel to R7 to change output current.
[image]

jpraus, 2020-05-04:
Excellent understatement of the board! Yeah, you are absolutely right.

> Great option.
> Q6 will definitely do the trick. Maybe R15 can have a bit higher value say 3.3V or 4.7K to reduce current consumption. But not higher than 12K
Theoretically yes, but I calculated the resistance to be 4.7K the same as you but when I tested it on b breadboard it does not work. I guess it's because of other resistance introduced by the breadboard. So 1K is temporary for now and I will be testing real value on the real device.
> 
> One more thing to add. As I can understand - Master Out is inverted.
> If we apply high logic level on Master out OT has ~6mA current
> With low level on Master out OT has ~20mA current
> 
Exactly, both OUT interfaces are inverted
> 
> So it is impossible to put jumpers between _master out_ -> _slave in_ and _slave out_ -> _master in_
> just to run this board without MCU attached. (in case of MCU failure or for test purposes)
> 
Yeas, it's impossible. It even does not make sense because if you want to run board without MCU - there is no need for the board in the first place. You can short the boiler and therm terminals.
> 
> Q1, Q2, R6 and R7 is a constant current generator. Like https://www.electroschematics.com/led-constant-current/
> Q3 in fact acts as a switch to add R9 in parallel to R7 to change output current.
> [image]
> 
Yes.
> 
> **Update:**
> This can be achieved by using MOSFET instead of BJT
> [image]
> Gate is isolated from 24V potential, so it is safe for MCU.
> Even if Gate shorts to the Drain a voltage divider 100K / 10K will reduce 24V to 2.182V wich is absolutely safe for MCU.
> 
That is great stuff. I had something similar in my mind but I did not want to introduce yet another component.