jpraus/arduino-opentherm #17: Understanding examples

Tommrodrigues, 2020-05-24:
Hey there, I'm having some difficulty understanding exactly how to use the library from the examples as they are quite limited. I have a couple of questions that should hopefully clarify some things for me:

1. Does the boiler need to be contacted every time in a loop, or can contact be event-based (e.g. when I want to change the temperature, change it then, rather than constantly setting it in each loop)
2. How can I set values? (e.g. the target temperature or the target central heating state)
3. How can I read values? (e.g. the upper and lower bound for heating, and the current boiler temperature)

I would really appreciate it if you could supply some examples as I'm having trouble back-calculating from the actual library source. Thanks!

jpraus, 2020-05-25:
I would recommend you reading the OpenTherm specification https://github.com/jpraus/arduino-opentherm/blob/master/doc/Opentherm%20Protocol%20v2-2.pdf. It will explain everything. Shortly:

1. Yes you need to contact boiler every 1 second a keep-alive signal.
2. You send setpoint temperature command
3. You send a command to read a certain value

For 2. and 3. you need to check the specification to find the appropriate command.

jpraus, 2020-05-25:
For the values you don't need to parse the upper and lower bound yourself. You can use utility functions on OpenthermData see https://github.com/jpraus/arduino-opentherm/blob/master/src/opentherm.h:

- f88
- u16
- s16

closed 2020-05-25 by jpraus