jpraus/arduino-opentherm #20: R8 on rev4 shield too low

JWGracht, 2020-11-11:
Correct me if I am wrong, because I am a digital specialist, not an analog.
But I think R8 on the rev4 shield must be 25k instead of 20k. Because the opentherm spec defines the voltage level low is up to 7V. The arduino spec for receiving a low is a voltage level below 1,5V. But when there is 7V at the THERM connector with R8 being 20K, there will be more then 1,5V at the MASTER-IN, so the arduino will not see the LOW signal.

mdvdl, 2020-12-24:
I had the same kind of problem. Had it looked at by two experts on electronics, so I replaced the R8 with a 27k. That works nicely.

Additionally, to get it working on a Wemos D1 Mini look-alike, I have not used the pin D3/ GPIO 0 for Boiler_out, but D5/GPIO 14. Reason is that D3 apparently  also has another function to "High run, low flash".

Now the test program is working fine.

jpraus, 2020-12-24:
Hi, thank you for your feedback. I am sure the 27K will not work for 5V logic since the high signal would be considered low. So I can recall there had to be some compromise in the margins. So let me retest it myself on my Wemos! Thank you!

niderost, 2021-01-01:
Possibly related: for the rev2. and rev3 boards, there is an instruction in the readme.md when using it in a 3.3V configuration:  Replace R11 resistor with 4k7 resistor (instead of 10k). Is this still applicable to the rev4 board?

jpraus, 2021-01-02:
> Possibly related: for the rev2. and rev3 boards, there is an instruction in the readme.md when using it in a 3.3V configuration: Replace R11 resistor with 4k7 resistor (instead of 10k). Is this still applicable to the rev4 board?

Hi! Not it's not. I need to update the documentation. To make the 3.3V work you just need to choose the 3.3V on logic voltage jumper.

IgorYbema, 2021-09-22:
I believe a 10v zener diode would help to prevent this (and remove R11 and lower R8 a bit). This will clamp down the idle voltage of max 7V and will pass through for a high level signal.