jpraus/arduino-opentherm #28: Rev 4 board with WeMos D1 R2 self-test failed

git1k2, 2021-02-14:
Hi, thanks for this fun project! I've just built some rev. 4 boards. And I'm testing with two WeMos D1 R2 boards and an Arduino Uno. For all boards the power supply test, interfaces test and self-test with Arduino Uno are OK.

The self-test with the WeMos D1 R2 boards and the shield plugged in, failed! No blinking LED's. The ESP was not starting, at least it was not running the program. I've also tested this with a simple blink program.

Using jumper wires I first powered on the WeMos. Then the self-test program starts running. Next I connected the the data lines and the test output was OK! So something was going on with the data lines being connected in combination with booting. When pushing the reset button (with all lines connected) the board would again not start the self-test.

I tried to debug the problem and think I found something. Would like to see if this makes sense. I tried moving BOILER_OUT from D3 to D4 and then to D5. With D5 it started to work!

```
Pins:
Uno,   Shield, WeMos D1 R2
7      7       D5 (gpio 14, works)
6      6       D4 (gpio 2, not ok)
5      5       D3 (gpio 0, not ok)
4      4       D2
3      3       D1
2      2       D0
1 tx   tx      tx
0 rx   rx      rx
```

Then I read this page: https://tttapa.github.io/ESP8266/Chap04%20-%20Microcontroller.html quote:

> ... some I/O pins have a special function during boot: They select 1 of 3 boot modes.

So GPIO15, **GPIO0** and **GPIO2** have a special function during boot. Would like to know if I'm headed in the right direction. I this is a known compatibility issue with the WeMos D1 R2 boards. Should I mod the board to use D5 (gpio 14) for BOILER_OUT?

mdvdl, 2021-02-16:
Hi Ben,

I had the same issue. I have connected the inputs as follows:

#define THERMOSTAT_IN D0 /*16 */
#define THERMOSTAT_OUT D2 /*4 */
#define BOILER_IN D1 /*5 */
#define BOILER_OUT D5 /*14 */

This is working fine. Hope it helps.

git1k2, 2021-02-17:
Hi @mdvdl thanks for confirming this issue. I see you also landed on D5 :)

jpraus, 2021-02-17:
Thank you for letting me know. I will consider changing the pinout.

git1k2, 2021-02-19:
Tried another board which seems pin compatible. The WeMos D1 R32 / WEMOS LOLIN32. It is a board with an ESP32.

```
// Wemos D1 R32
#define THERMOSTAT_IN 26
#define THERMOSTAT_OUT 17
#define BOILER_IN 25
#define BOILER_OUT 16
```

```
OpenTherm gateway self-test
Boiler inbound, thermostat outbound .. OK
Boiler outbound, thermostat inbound .. OK
...

```

pavelhrolovich, 2023-01-07:
I can confirm the same issue - switch to D5 worked just fine.

xnosek00, 2024-01-07:
Hi all, I can confirm the same problem with Wemos D1 R2. The device didn't boot with connected Rev 4 board and connected PIN D3. 

Switch to D5 works. 

@jpraus it cost me time, before I found the problem and this issue. Discussion in this issue helped. Is it possible to write this information to main documentation. I think, it could help more people and it could save their time.