jpraus/arduino-opentherm #34: Why is the value of R5 10k?

NijzinkM, 2021-05-04:
Thank you for sharing these schematics! I built the gateway myself with Wemos D1 mini and it has been running successfully for a year now, apart from one issue.

About half the time when the gateway restarts, it locks itself. SLAVE IN stays high and MASTER IN stays low permanently. The thermostat doesn't boot. Turning the gateway off and on a few times resolves the issue.

I think it might have something to do with R5. Why does it have a value of 10k and not 1.5k for example? Lowering the value of R5 increases the difference betwoon LOW and HIGH on SLAVE IN. The 10k value might cause the SLAVE IN value to float between HIGH and LOW.

Simulation of the boiler interface: [shorturl.at/LWY06](url)

Can you explain why you chose 10k for R5?

jpraus, 2021-05-05:
Hi! Thank you for your feedback.
I believe your simulation is wrong. The R5 is a pull-down resistor for the input pin of the MCU. The R5 in simulation works as a voltage divider, which is not true when connected to MCU GPIO.

jpraus, 2021-05-05:
What does it mean that the gateway locks itself?

NijzinkM, 2021-05-06:
Thank you for the explanation. Out of curiosity I lowered R5 to 1.5k, which is low for a pull-down resistor but not problematic. It may have something to do with the optocouplers I'm using. I will let you know how it works out.

By locking I mean that the MCU doensn't boot properly. It is powered but it doesn't send any data or response. GPIO's 0, 1, 2 and 15 affect the boot process of the ESP8266, but I'm not using those pins. I use 4, 5, 12 and 14. This might be a Wemos D1 issue.

You've answered my question about R5 so you can close the issue if you want.

closed 2022-02-13 by NijzinkM