jpraus/arduino-opentherm #36: Support for opentherm smartpower

IgorYbema, 2021-09-22:
Opentherm smartpower allows thermostats to ask more power from the slave. This is not in the code yet.
It uses same schematic and is only software based. In fact, it is just swapping the low/high logic where on high power both voltage and ampere are at idle high (for medium power only amps are at idle high). At least that is what I understand from the original opentherm gateway designer.

Was looking at the library how to implement this but I got a bit puzzeled. So maybe just a quick ask here if you might be interested in implementing it.

KEYvv, 2023-01-26:
Hi.
I'm into Opentherm the smart power and looking for reference code for a certification.
Are you still making a code for Opentherm smartpower?

IgorYbema, 2023-01-26:
I have got some simple implementation but it doesn't work flawless yet. This library isn't compatible enough with the different states. And I don't have the opentherm 3 specification where smart power is explained so I am doing it from knowledge I gain from others. Do you have the smart power specification?

IgorYbema, 2023-01-26:
You can find some addtions to the libary in this version. You can spot the smartpower changes. https://github.com/IgorYbema/HeishaMon/blob/opentherm/HeishaMon/OpenTherm.cpp

Phunkafizer, 2025-04-02:
I tried to implement smartpower here:
https://github.com/Phunkafizer/opentherm_library
As I do not have the open therm spec. 3 it's just "reengeneering" & trying, but it seems to work well at least with the Honeywell DT41SP thermostat. If anyone has a smartpower roomunit I would be happy to get feedback.