jpraus/arduino-opentherm #37: Wemos D1 R1, can't configure pin D2 with pull up

halecivo, 2022-01-16:
Hello, I use the shield with wemos d1 r1. 

Shield uses pins D2, D3, D4 and D5, which is mapped to gpio 16, 5, 4, 14 according to [image]

I use following mapping taken from an example:
```
    static constexpr uint8_t THERMOSTAT_IN = 16;
    static constexpr uint8_t THERMOSTAT_OUT = 4;
    static constexpr uint8_t BOILER_IN = 5;
    static constexpr uint8_t BOILER_OUT = 14;
```

And following initialization of pins:

```
        pinMode(THERMOSTAT_IN, INPUT);
        digitalWrite(THERMOSTAT_IN, HIGH); // pull up
        digitalWrite(THERMOSTAT_OUT, HIGH);
        pinMode(THERMOSTAT_OUT, OUTPUT); // low output = high current, high output = low current
        pinMode(BOILER_IN, INPUT);
        digitalWrite(BOILER_IN, HIGH); // pull up
        digitalWrite(BOILER_OUT, LOW);
        pinMode(BOILER_OUT, OUTPUT); // low output = high voltage, high output = low voltage
```

I noticed that once I initialize GPIO16 this way, OTA function of esphome stops working. I checked [esphome](https://esphome.io/devices/esp8266.html) page to see that GPIO16 can be configured to pulldown only, [here](https://www.esp8266.com/viewtopic.php?p=90211) it seems confirmed as limitation of ESP8266.

The opentherm communication worked well, but I need OTA functionality as well to be able to update firmware without dismantling the system. 

1. Could somebody with deeper experience confirm me source of this problem?
2. Would external pull up solve the issue (sources tells that GPIO16 can't be configured with pull up, not that pull up can't be used at all)?
3. Opentherm is not polarity sensitive - could swapping THERMOSTAT IN and OUT pins solve the issue?
4. Is there a different pin I could rewire THERMOSTAT_IN pin to?

Thanks.

jpraus, 2022-01-18:
Hi! Thanks for an elaborative explanation of your problem! That really helps!
I have no experience with esphome so I don't know what is the story around GPIO16. But for your options:
2. I don't think so, moreover you would break the functionality of the shield
3. It's polarity insensitive on the input, the output pins are "polarity sensitive"
4. Yes you can use any pin as long as it's free to use. The library is not using interrupts so it can by any IN/OUT pin

NijzinkM, 2022-02-13:
I use pin 12 instead of 16 on my ESP8266 and OTA is working fine! I also use the pins in a different order, because of the layout of my circuit. As jpraus said, you can use any pin as long as it's not used by something else. [https://randomnerdtutorials.com/esp8266-pinout-reference-gpios/](https://randomnerdtutorials.com/esp8266-pinout-reference-gpios/)